TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #1: Klipper Neptune3Pro config - filament sensor

jfurner, 2023-01-26:
Hey just a small update for the Klipper N3P config.  The filament sensor type appears to be a switch instead of a motion sensor.  It kinda works as-is but it usually pauses the print right at the beginning for a false trigger filament runout.  Then it seems to work after that, which I'm not sure I understand why it doesnt just keep triggering filament runout.

I replaced section [filament_motion_sensor filament_sensor] with this, and no more pauses at the beginning and the filament sensor state on the dashboard updates correctly.

[filament_switch_sensor filament_sensor]
pause_on_runout: true
switch_pin: PB4

Thanks again for all your work on this!  Its working great.

TheFeralEngineer, 2023-01-26:
Thanks for the update! The original cfg was built around the Neptune 3 standard file, so I'll be sure to change the one for the pro. Glad to hear it's working otherwise 😀

mroswald, 2023-10-10:
I have the same issue on my Neptune 3 (not pro!). Will change the config and test it with the `filament_switch_sensor` instead of `filament_motion_sensor`.

TheFeralEngineer, 2023-10-10:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/1/1755260477
> @github.com>

You need the motion sensor setting on the 3 standard or it'll give you
issues

mroswald, 2023-10-10:
Sorry for the noise then!

TheFeralEngineer, 2023-10-10:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/1/1755528243

If you're having issues on the 3, take the sensor apart and check for
buildup inside of it and/or a stuck encoder wheel

mroswald, 2023-10-10:
I could finally make it work by increasing `detection_length` (default 7mm, I've upped to 50mm) and using the `filament_motion_sensor` (the settings you provided). Thanks!