TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #2: Neptune 3 Plus Config Changes

BongoEADGC6, 2023-02-02:
In case anyone is looking for answers in this repo. 

The communications settings for the Neptune 3 Pro firmware doesn't work for the Neptune 3 Plus. In order for the Neptune 3 Plus to operate it must use USART2 PA3/PA2 for connectivity.

TheFeralEngineer, 2023-02-02:
Good to know! I'll pop out a bin file for it asap

BongoEADGC6, 2023-02-03:
Only problem at the moment is figuring out the printer.cfg regarding the axes and offsets. Having a difficult time getting those solved at the moment.

jdoncouse, 2023-02-09:
I have the N3Pro USB firmware loaded right now on my plus and it's working just as well as my pro as far as i can tell. 

I did have the issue described in this post.
https://www.reddit.com/r/klippers/comments/ybqvse/trouble_connecting_to_mcu_mks_gen_l_10/

sudo systemctl stop brltty-udev.service
sudo apt remove brltty
fixed the issue on a fresh install of Pop!_OS

3Dtcb, 2023-03-21:
> 

Any chance to get the updated firmware for the N3Plus?

TheFeralEngineer, 2023-03-21:
> > 
> 
> Any chance to get the updated firmware for the N3Plus?

The bin file should work with all models of Neptune 3

3Dtcb, 2023-03-21:
> > Any chance to get the updated firmware for the N3Plus?
> 
> The bin file should work with all models of Neptune 3

I thought you said that a new bin file is needed in february?

3Dtcb, 2023-03-24:
If someone else reads this ... I found the solution because I made 2 mistakes.

1. Use the "Serial UART communication" firmware for the Neptune 3 Plus
2. Use the upper usb port of the sonic pad (RTF! I know ...)

Maybe this information will avoid stupid questions like mine in the future.

closed 2023-11-14 by BongoEADGC6