TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #8: Neptune 3 Plus

lewis2e, 2023-03-24:
The plus has 6 screws, while the configuration file only defines 4. I am 100% brand new to Klipper, so please accept my apologies if this isn't actually an issue.

TheFeralEngineer, 2023-03-24:
> The plus has 6 screws, while the configuration file only defines 4. I am 100% brand new to Klipper, so please accept my apologies if this isn't actually an issue.

These are early stage configs and I don't have a plus/max to test, so I'm hoping to have some people fine tune them and do a pull request to bring them up to snuff.

Additional bed points can be added to printer.cfg

lewis2e, 2023-03-24:
Thanks for the reply, and I appreciate all your efforts. This became a moot point for me anyway as I couldn't even flash the firmware (neither the bin you provide in this repo, not the bin I compiled myself). Had to give up unfortunately.

lewis2e, 2023-03-24:
No. I feel like I've read everything and I've never seen any reference to disconnecting the screen. I've seen people talk about how you can't use the screen, but nothing about disconnecting it. I'll give that a go.

lewis2e, 2023-03-24:
I tried this and it did not make a difference. I disconnected the screen and booted with the SD card in with the correctly named bin file. I can see from the Raspberry Pi that the connection becomes listed under /dev/serial, but I still cannot connect. Did several RESTARTs from the Octoprint console, all resulting the in the same "Printer not ready" error. Taking the SD out later, I can see that the bin file has been renamed to .cur, but I don't know if that is conclusive of a successful flash or not. Not sure what I am doing wrong.

Anyway - I'm not expecting or demanding support from you. Happy to provide more info or try more things if it's helpful - but I'm miles away my original topic now!

3Dtcb, 2023-03-24:
You have to use the "Serial UART communication"  firmware and you have to use the upper USB port of the Sonic Pad.  
These are the 2 things I messed up. ;)

3Dtcb, 2023-03-24:
>  The biggest problem I've run into with these is people not disconnecting the screen before flashing the bin. Did you try that?

Do you mean the screen of the Neptune or the Sonic Pad?
I flashed the printer with the Neptune screen and it is working.
I tried to remove the screen but then the Sonic Pad stops working.
I assumed that the original screen needs to be attached. It always shows "Firmware Update" and the Elegoo Logo but everything is working

lokuras87, 2023-03-27:
I hope it will be useful to someone, it is what I use in my NP3 Plus, I am Spanish and I use google to translate, I don't know very well how github works hehe

[bed_mesh]
speed: 150
horizontal_move_z: 5.0
mesh_min: 10, 25
mesh_max: 291.5, 276
probe_count: 6,6
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0

[screws_tilt_adjust]
screw1: 39, 35
screw2: 39, 167
screw3: 39, 300
screw4: 295, 300
screw5: 295, 167
screw6: 295, 35

3Dtcb, 2023-03-27:
@lokuras87 Thanks! I will try your settings after my print is finished!

lewis2e, 2023-03-27:
I got this working in the end. I started with the config file in this repository. These are the bits I changed:

[bed_mesh]
speed: 200
horizontal_move_z: 5
mesh_min: 22,22
mesh_max: 290,315
probe_count: 7,7
fade_start: 1.0
fade_end: 10.0
algorithm: bicubic

[screws_tilt_adjust]
screw1: 65, 13
screw1_name: left front screw
screw2: 65, 145
screw2_name: left middle screw
screw3: 65, 280
screw3_name: left back screw
screw4: 318, 13
screw4_name: right front screw
screw5: 318, 145
screw5_name: right middle screw
screw6: 318, 280
screw6_name: right back screw

[safe_z_home]
home_xy_position: 160, 160
z_hop: 10

[gcode_macro M600]
description: Pause to change filament colour
gcode:
  PAUSE

3Dtcb, 2023-03-28:
@lewis2e Thanks! I tried your old settings today and I got only 5 out of 6 mesh points working. 
I just started a print and will try your new values tomorrow.

I added the following to my cfg to load and unload filament with a macro.
Not sure if there's a better way to do it! :)

```
[gcode_macro LOAD_FILAMENT]
#
# LOAD_FILAMENT
#
# Heat to 200°C then prime the nozzle
description: Load filament (heat nozzle if necessary)
gcode:
   M117 Loading filament...
   {% if printer[printer.toolhead.extruder].temperature < 200.0 %}
   STATUS_HEATING
   M117 Heating hotend...
   SET_HEATER_TEMPERATURE HEATER='extruder' TARGET=200
   TEMPERATURE_WAIT SENSOR='extruder' MINIMUM=200
   {%endif%}
   STATUS_BUSY
   M117 Loading filament...
   M83                            ; set extruder to relative
   G1 E40 F300                    ; slower extrusion for hotend path
   G1 E15 F150                    ; prime nozzle with filament
   M82                            ; set extruder to absolute
   UPDATE_DELAYED_GCODE ID=_CLEAR_DISPLAY DURATION=10
   STATUS_READY

[gcode_macro UNLOAD_FILAMENT]
#
# UNLOAD_FILAMENT
#
# Heat the nozzle to 230°C then retract the filament out of the toolhead
#
description: Heat the nozzle (if needed), then retract the filament out of the toolhead
gcode:
  {% if printer[printer.toolhead.extruder].temperature < 230.0 %}
    STATUS_HEATING
    M117 Heating hotend...
    SET_HEATER_TEMPERATURE HEATER='extruder' TARGET=230
    TEMPERATURE_WAIT SENSOR='extruder' MINIMUM=230
  {% endif %}
  STATUS_BUSY
  M117 Unloading filament...
  M83                            ; set extruder to relative
  G1 E2 F300                    ; extrude a little to soften tip
  G1 E-80 F1800                 ; retract filament completely
  M82                            ; set extruder to absolute
  UPDATE_DELAYED_GCODE ID=_CLEAR_DISPLAY DURATION=10
  STATUS_READY

```

3Dtcb, 2023-03-30:
@lewis2e WOW! Thanks for your settings. It looked perfect when I leveled the bed!

rod419, 2023-06-08:
I'm fairly new to Klipper, so please excuse my ignorance, but is adding the code that @lewis2e provided supposed to show me all 6 screw points in the "manual Level" menu? 

I added the code but I'm still only able to hit 4. The menu shows all 4 corners, but they correspond to 1 - lower, 2 - mid and 3 - top of the left side, and 4 - Bottom right. 

is there another setting i should check?

TheFeralEngineer, 2023-06-08:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/8/1583583546
> @github.com>

That's a limitation in KlipperScreen if you're using that. You'll need to
do the screw adjust through webui