TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #9: Neptune 3 Plus - Bed Leveling

3Dtcb, 2023-03-24:
When I level the 3 Plus then it doesn't measure the right side and the back of the bed. 
I am not sure why and I checked the printer.cfg and I may found the problem.

All config have these settings:

```
[bed_mesh]
speed: 300
horizontal_move_z: 5.0
mesh_min: 30,30
mesh_max: 205,205
probe_count: 6,6
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0
```

The bed of the Neptune 3 Pro is 225 x 225 x 280 mm
The bed of the Neptune 3 Plus is 320 x 320 x 400 mm
The bed of the Neptune 3 Max is 420 x 420 x 500 mm

Should the mesh_max change when the print bed gets bigger?
Plus: mesh_max: 305,305
Max: mesh_max: 405,405

I am really a beginner in FDM printing but I assume this might be an error.
I am too afraid to change something. ;)

My assumption is based on https://www.klipper3d.org/Bed_Mesh.html

3Dtcb, 2023-03-25:
I tried it with 305,305 and it didnt work. 
Somehow the printer wasnt able to measure at 305.

```
00:52:46 
$ BED_MESH_CALIBRATE
00:53:53 
//auto_bed_level 1,58.000,10.000,0.050000
00:53:54 
//auto_bed_level 2,113.000,10.000,0.057500
00:53:56 
//auto_bed_level 3,168.000,10.000,0.090000
00:53:57 
//auto_bed_level 4,223.000,10.000,0.122500
00:53:59 
//auto_bed_level 5,278.000,10.000,0.117500
00:53:59 
!! {"code":"key243","msg":"Move out of range: 333.000 10.000 5.000 [0.000]", "values":[333.000, 10.000, 5.000, 0.000]}
```

I am not sure why it is always moving 55 

305 (max) - 30 (min) = 275
275 / 6 (probe count) = 45
275 / 5 = 55 

It should move 45 and not 55.
Am I missing something? Do you know what I mean?

lewis2e, 2023-03-30:
@3Dtcb - I think you are all sorted now but, for the sake of anyone else coming across this post, the issue with "Move out of range" is because Klipper will minus the probe offset before moving to that position.

For example, given the 320x320 bed, you might think you can put `mesh_max: 300,300` into the config file. However, to get the probe to 300,300, it will try to move the nozzle to 328,280 (i.e. 300-(-28), 300-(20). Obviously X328 is out of bounds.

I've given (what I believe to be) a workding [bed_mesh] section here: - https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/8

I personally think this is confusing. In the` [bed_mesh]` section you are expected to enter the _**probe**_ coordinates and Klipper determines the **_nozzle_** coordinates for itself. In contrast, in the `[screws_tilt_adjust]` section you are expected to enter the **_nozzle_** coordinates and Klipper determines the **_probe_** coordinates for itself... the reverse.

Maybe it's just me (I'm very new to Klipper) but that seems a bit silly.

mastertape, 2023-04-16:
ich habe ein ähnliches Problem beim Neptune 3 pro...

interessanter weise hat die cfg file am sonic pad sofort funktioniert... als ich jetzt mittels  kiauh auf der raspi 3 das ganze aufgestzt habe beobachte ich selbiges Phänomen. Allerdinghs ist es bei meinen 3 Pro der linke und obere bereich der im auto leveling prozess ausgelassen wird.

inlfwetrust, 2024-10-20:
Not sure if it is the same issue, I attached my config file, which I downloaded from here
[config-20241021-080922.zip](https://github.com/user-attachments/files/17452256/config-20241021-080922.zip)
The issue is it starts to level my bed not from the left edge but~3-5 cm from the left, but from the right side it measures almost over the edge.
Is it a feature or do I miss some settings?

TheFeralEngineer, 2024-10-20:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/9/2425257490
> @github.com>

That's because it's lining up with the sensor, not the nozzle. Completely
normal.
Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc