TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #11: Orange Pi LTS 3 or Orange Pi LTS 4 + Screen

Piero87, 2023-04-12:
Hello,
Thank you very much for the work you made, I have some questions:

1. Can I use Orange Pi LTS 3 or Orange Pi LTS 4 to install Klipper?
2. Can I connect a screen to the Orange Pi to see the Klipper interface like the Sonic Pad of Creality?
3. if i successfully install klipper and connect it to the usb port of the printer, can i then disconnect the original screen?
4. What is the difference between this project and the file already inside Klipper repository (https://github.com/Klipper3d/klipper/blob/master/config/printer-elegoo-neptune3-pro-2023.cfg)

Thank you again!

TheFeralEngineer, 2023-04-13:
Hello 😺 I hope these help answer your questions: 1) yes, it will work with Klipper2) I have not used an Orange Pi personally, so I can’t guarantee that a screen will work. I know someone that is having issues with one working right now. I would suggest the Bigtreetech Pi4b and CB1 listed here: https://shrsl.com/3yjb13) with the Pi4B/CB1, you can choose to run either from USB or UART (I have a small video on how to do it with a Makerbase MKS Pi, but the steps are similar for CB1. I will do a video on it in the future to show the differences). You need to disconnect the original screen in order for the ZNP_ROBIN_NANO.bin file to load on the printer’s motherboard4) the difference is that I made the printer.cfg on my GitHub page, someone else made the one you sent the link for.  From: Piero87Sent: Wednesday, April 12, 2023 2:27 AMTo: TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-PrintersCc: SubscribedSubject: [TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers] Orange Pi LTS 3 or Orange Pi LTS 4 + Screen (Issue #11) Hello,Thank you very much for the work you made, I have some questions:Can I use Orange Pi LTS 3 or Orange Pi LTS 4 to install Klipper?Can I connect a screen to the Orange Pi to see the Klipper interface like the Sonic Pad of Creality?if i successfully install klipper and connect it to the usb port of the printer, can i then disconnect the original screen?What is the difference between this project and the file already inside Klipper repository (https://github.com/Klipper3d/klipper/blob/master/config/printer-elegoo-neptune3-pro-2023.cfg)Thank you again!

Piero87, 2023-04-13:
Thank you very much, but I have still to install the firmware that there is in your repository?

closed 2023-06-18 by Piero87