TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #12: Neptune 3 Pro issue with auto bedleveling calibration

mastertape, 2023-04-17:
I have a similar problem with the Neptune 3 pro... like the other issue reportet with the 3 Plus

interestingly, the cfg file worked immediately on the sonic pad... when i used kiauh on the raspi 3 klipper, i observed the same phenomenon. However, with my 3 Pro it is the left and upper area that is left out in the auto leveling process.

Incidentally, I have adopted changes to the config file I use on the sonic pad (of course taking into account device-specific changes such as gcode opath directory serial port etc.)...

I assume that the cfg file is correct in terms of the configuration and values. where else could the problem be? where is the bed size asked at klipper? If you added a new printer to the sonic pad, the print bed size had to be specified once.... perhaps the solution lies in entering these values... which additional config file could this affect?

here is also my printer.cfg

`# For the ZNP Robin Nano DW v2.2 board:
#   - Compile with the processor model STM32F401.
#   - Select the 32KiB bootloader,
#   - Select (Serial (on # USART1 PA10/PA9) for the communication interface.
#   - Select (Serial (on # USART2 PA3/PA2) if you want to use the serial UART pins on J17 (wifi section)
# Note that the "make flash" command does not work with ZNP Robin boards. After running "make", run the following command:
# cp out/klipper.bin out/ZNP_ROBIN_NANO.bin
#
# Copy the file out/ZNP_ROBIN_NANO.bin to an SD card and then restart the printer with that SD card.
# See docs/Config_Reference.md for a description of parameters.

#[include timelapse.cfg]

[exclude_object]
[virtual_sdcard]
path: /home/pi/printer_data/gcodes
#path: /mnt/UDISK/.crealityprint/upload //for sonic pad
[pause_resume]

[display_status]

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
    TURN_OFF_HEATERS
    CANCEL_PRINT_BASE

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
# change this if you need more or less extrusion
variable_extrude: 1.0
gcode:
    ##### read E from pause macro #####
    {% set E = printer["gcode_macro PAUSE"].extrude|float %}
    ##### set park positon for x and y #####
    # default is your max posion from your printer.cfg
    {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
    {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
    ##### calculate save lift position #####
    {% set max_z = printer.toolhead.axis_maximum.z|float %}
    {% set act_z = printer.toolhead.position.z|float %}
    {% if act_z < (max_z - 2.0) %}
        {% set z_safe = 2.0 %}
    {% else %}
        {% set z_safe = max_z - act_z %}
    {% endif %}
    ##### end of definitions #####
    PAUSE_BASE
    G91
    {% if printer.extruder.can_extrude|lower == 'true' %}
      G1 E-{E} F2100
    {% else %}
      {action_respond_info("Extruder not hot enough")}
    {% endif %}
    {% if "xyz" in printer.toolhead.homed_axes %}
      G1 Z{z_safe} F900
      G90
      G1 X{x_park} Y{y_park} F6000
    {% else %}
      {action_respond_info("Printer not homed")}
    {% endif %} 
    
[gcode_macro RESUME]
description: Resume the actual running print
rename_existing: RESUME_BASE
gcode:
    ##### read E from pause macro #####
    {% set E = printer["gcode_macro PAUSE"].extrude|float %}
    #### get VELOCITY parameter if specified ####
    {% if 'VELOCITY' in params|upper %}
      {% set get_params = ('VELOCITY=' + params.VELOCITY)  %}
    {%else %}
      {% set get_params = "" %}
    {% endif %}
    ##### end of definitions #####
    {% if printer.extruder.can_extrude|lower == 'true' %}
      G91
      G1 E{E} F2100
    {% else %}
      {action_respond_info("Extruder not hot enough")}
    {% endif %}  
    RESUME_BASE {get_params}

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

#[led LED_Light]
#white_pin: PB9
#initial_white: 1.0

[output_pin LED_Light] #toggle for LED Light - use this instead of [LED] for sonic pad
Pin: PB9
cycle_time: 0.01
pwm: true
value: 1

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 3000
max_z_velocity: 15
max_z_accel: 100
square_corner_velocity: 5
# Use those higher values just to configure Input Shaper 
#max_accel: 10000
#max_accel_to_decel: 10000

[stepper_x]
step_pin: !PC12
dir_pin: PB3
enable_pin: !PD2
microsteps: 16
rotation_distance: 40
endstop_pin: PA13
position_endstop: -6.0
position_min: -6
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC11
dir_pin: PA15
enable_pin: !PC10
microsteps: 16
rotation_distance: 40
endstop_pin: PB8
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PC7
dir_pin: !PC9
enable_pin: !PC8
rotation_distance: 8
microsteps: 16
position_min: -2
position_max: 280
endstop_pin: probe:z_virtual_endstop # Use Z- as endstop
homing_speed: 5.0

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB10
dir_pin: PB1
enable_pin: !PC6
microsteps: 16
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA6
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC1
min_temp: 0
max_temp: 250
# Calibrate E-Steps https://www.klipper3d.org/Rotation_Distance.html#calibrating-rotation_distance-on-extruders
rotation_distance: 6.9
# Calibrate PID: https://www.klipper3d.org/Config_checks.html#calibrate-pid-settings
#  - Example: PID_CALIBRATE HEATER=extruder TARGET=200
control = pid
pid_kp = 30.356
pid_ki = 1.857
pid_kd = 124.081
# Calibrate PA: https://www.klipper3d.org/Pressure_Advance.html
#pressure_advance = 0.600

[heater_bed]
heater_pin: PA5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
pwm_cycle_time: 0.020 # set to 0.0166 if your grid runs on 60Hz to fix lights flickering
max_temp: 110
min_temp: 0
# Calibrate PID: https://www.klipper3d.org/Config_checks.html#calibrate-pid-settings
#  - Example: PID_CALIBRATE HEATER=heater_bed TARGET=60
control = pid
pid_kp = 64.230
pid_ki = 0.723
pid_kd = 1425.905

[heater_fan hotend_fan]
pin: PB0
heater: extruder
heater_temp: 50.0

[fan]
pin: PA7

[force_move]
enable_force_move: True

[safe_z_home]
home_xy_position: 117.5, 117.5
z_hop: 10

[probe]
pin: ^PA8
speed: 5
lift_speed: 15
samples: 1
x_offset: -28
y_offset: 20
# Calibrate probe: https://www.klipper3d.org/Bed_Level.html
# - Example: PROBE_CALIBRATE, then adjust with TESTZ Z=+/-X
#z_offset = -0.100

[filament_switch_sensor filament_sensor]
pause_on_runout: true
switch_pin: PB4

[bed_mesh]
speed: 300
horizontal_move_z: 5.0
mesh_min: 30,30
mesh_max: 205,205
probe_count: 6,6
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0

[input_shaper]
# Calibrate IS: https://www.klipper3d.org/Resonance_Compensation.html
#shaper_type_y = mzv
#shaper_freq_y = 38.2
#shaper_type_x = ei
#shaper_freq_x = 79.6

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
#min_temp: 10
#max_temp: 105

[temperature_sensor mcu_temp]
sensor_type: temperature_mcu
sensor_temperature1: 25
sensor_adc1: 0.210317
#min_temp: 0
#max_temp: 105

[gcode_macro M420]
description: Load the current mesh
gcode:
  BED_MESH_PROFILE LOAD=default
  
[gcode_macro G29]
description: creates automated homing and bed mesh
gcode:
  G28
  BED_MESH_CALIBRATE
  DATA_SAVE

# [mcu rpi]
# serial: /tmp/klipper_host_mcu

# [adxl345]
# cs_pin: rpi:None
# spi_speed: 2000000
# spi_bus: spidev2.0

# [resonance_tester]
# accel_chip: adxl345
# accel_per_hz: 70
# probe_points:
#     150,150,10

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe]
#*# z_offset = 2.065
#*#
#*# [input_shaper]
#*# shaper_type_x = ei
#*# shaper_freq_x = 74.4
#*# shaper_type_y = zv
#*# shaper_freq_y = 41.8
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  -0.070000, 0.025000, 0.135000, 0.225000, 0.345000, 0.435000
#*# 	  -0.157500, -0.037500, 0.070000, 0.152500, 0.275000, 0.330000
#*# 	  -0.137500, -0.037500, 0.060000, 0.127500, 0.220000, 0.262500
#*# 	  -0.152500, -0.062500, 0.020000, 0.087500, 0.170000, 0.215000
#*# 	  -0.120000, -0.045000, 0.020000, 0.072500, 0.147500, 0.180000
#*# 	  -0.090000, -0.027500, 0.037500, 0.095000, 0.160000, 0.210000
#*# x_count = 6
#*# y_count = 6
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 30.0
#*# max_x = 205.0
#*# min_y = 30.0
#*# max_y = 205.0
`

iHunterX, 2023-05-22:
I second it, I have the same issue, the first layer always seem too high

TheFeralEngineer, 2023-05-22:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/12/1557792928
> @github.com>

When setting z offset, do you command probe_calibrate or just try and set
it from the axis controls?

Meckimesser75, 2024-02-15:
Hi there,

having the same Problem here.

Ist there any update/solution for this?

Working whith Marlin everthing ist perfekt.

Thx a lot for support