TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #16: 4 Tilt-Adjust Screws in Neptune 3 Max config instead of 6

adam120800, 2023-06-02:
There should 6 Tilt-Adjust Screw defined in Neptune 3 Max cfg like on the Plus printer.cfg. If anyone has a Max version with all 6 measured screws location please reply.

TheFeralEngineer, 2023-06-02:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/16@
> github.com>

Easy enough to add the other two screws. Just find the screw section and
add screw5: and screw6:. I don't have one, so I can only offer up what i
have. I'm waiting on a pull request so i can update my config files 🤷‍♂️

adam120800, 2023-06-02:
Found this from a fellow reddit post 13hours ago you commented on aswell.  This should work for N3 Max:

[screws_tilt_adjust]
screw1: 40, 40

screw2: 40, 230

screw3: 40, 400

screw4: 400, 400

screw5: 400, 230

Screw6: 400, 40

TheFeralEngineer, 2023-06-02:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/16/1574334218

Sweet, thanks! I'll fix the max file as soon as I get a chance 🙂

closed 2023-06-26 by TheFeralEngineer