TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #20: Bed Level Problem

Piero87, 2023-06-22:
Hi, I have a print problem with my Elegoo Neptune 3 Pro with Klipper, after all the configuration, Bed mesh calibration and probe calibration, z offset with the paper, when start printing on the first layer does not stick well to the bed, before switching to klipper I never never had 1 problem with this 3d printer, anyone can help me?

here is the calibration part of the .cfg:

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 24.041
#*# pid_ki = 1.418
#*# pid_kd = 101.875
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 65.198
#*# pid_ki = 0.783
#*# pid_kd = 1356.934
#*#
#*# [probe]
#*# z_offset = 2.315
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*#     0.520000, 0.492500, 0.475000, 0.432500, 0.415000, 0.427500
#*#     0.360000, 0.302500, 0.292500, 0.290000, 0.285000, 0.302500
#*#     0.230000, 0.167500, 0.160000, 0.157500, 0.147500, 0.145000
#*#     0.050000, -0.005000, -0.007500, 0.002500, 0.010000, 0.030000
#*#     -0.080000, -0.142500, -0.152500, -0.157500, -0.147500, -0.125000
#*#     -0.222500, -0.267500, -0.282500, -0.280000, -0.257500, -0.230000
#*# x_count = 6
#*# y_count = 6
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 10.0
#*# max_x = 205.0
#*# min_y = 25.0
#*# max_y = 220.0

This is my bed mesh:
[image]

aaaaahaaaaa, 2023-06-24:
Exact same issue here

TheFeralEngineer, 2023-06-24:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/20/1605582829
> @github.com>

Did you use probe_calibrate to set your z offset? If you just used the
controls in the webui, it'll never come out right

aaaaahaaaaa, 2023-06-24:
@TheFeralEngineer Yep, the z offset was calibrated with `probe_calibrate`. What I just noticed is that I have to calibrate it lower with Klipper than I used to with the default `Elegoo` firmware, as in, using a tighter "paper test" pressure. I did a couple of prints and that seems to hold now.

Piero87, 2023-06-25:
Yes I use the command PROBE_CALIBRATE and with mainsail appear the dialog with the command to set it

TheFeralEngineer, 2023-06-26:
I haven't had this issue, unfortunately. I usually run bed_mesh_calibrate, restart, load the mesh, home the machine to the center of the bed, run probe_calibrate and everything is right as rain.