TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #21: neptune 3 plus x,y positioning

Craigeous, 2023-06-27:
i'm having some issues with positioning on my bed. it seems like the x axis is off a bit. when i start prints and it draws the first line it's not fully on the bed. it seems to hit the end stop and stay there. i didn't have this issue with the default firmware on the machine. i modified the stepper_x min position and that didn't seem to do anything either. i'm just getting into klipper can i cannot find any docs that speak to how to adjust this. the screw positions seem to be off for me. the front left screw is 40,35 and the center of the build plate is around 168,165 for me. these coordinates is with me using the default settings of the n3plus printer.cfg file from the repo. not sure what i need to do to tune this correctly, but doesn't seem like i can use klipper with my printer in this state.

lokuras87, 2023-06-27:
This configuration works well for me, I got it through trial and error.

[bed_mesh]
speed: 150
horizontal_move_z: 5.0
mesh_min: 10, 25
mesh_max: 291.5, 276
probe_count: 6,6
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0

[screws_tilt_adjust]
screw1: 39, 35
screw2: 39, 167
screw3: 39, 300
screw4: 295, 300
screw5: 295, 167
screw6: 295, 35

Craigeous, 2023-06-28:
that's a little better. the first line is right on the edge for the build plate now. so increasing the mesh_min x value will push the limit for x to the left? i guess my understanding was that bed_mesh was for leveling.

TheFeralEngineer, 2023-06-28:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/21/1611898034
> @github.com>

Bed mesh isn't for your x and y limits. You need to change the
position_endstop and position_min from -6 to -10 or -12. That should fix
the issue

Phil T.
The Feral Engineer

Craigeous, 2023-06-28:
ok... that's what i was wondering because i thought bed_mesh was for leveling. i'll give that a shot. the other thing i noticed is the print quality is noticeably worse. like rounded objects seem to be lower resolution or something along those lines. it's so strange.

rougewan, 2023-07-29:
Hey @Craigeous do you mind sharing your config for your plus.. I am not sure mine is correct and wanted to check mine with yours.