TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #22: Neptune 3 pro .bin update problem

pattes, 2023-06-28:
Hi TheFeralEngineer,

Sorry but the firmware has a problem for the neptune 3 pro. The update is stuck on firmware update. Would you be so kind as to update the firmware according to the official firmware released in June.

Thank you and see you soon!

TheFeralEngineer, 2023-06-28:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/22@
> github.com>

Disconnect the screen and try again. That's the biggest reason why it won't
update on most systems. I uploaded a new klipper firmware bin 3 days ago.

Phil T.
The Feral Engineer

TheFeralEngineer, 2023-06-28:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/22/1611727554

Anytime. Good luck! 🐱🐱

Phil T.
The Feral Engineer

Dedoque, 2023-11-21:
Also can you provide `.config` for you build?

franzs, 2023-12-04:
@Dedoque The options for Klipper's `make menuconfig` are mentioned here:

https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/blob/5c1ae03770af8fcbae772870ec8d73267b088dd6/Neptune%203%20Pro%20config/printer.cfg#L1-L10

This is my `.config` for a Neptune 3 Pro using Klipper v0.12.0 communicating via USB: 
[.config](https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/files/13551824/default.config.txt)

Make sure to name this file `.config`.

zloyded, 2023-12-05:
> @Dedoque The options for Klipper's `make menuconfig` are mentioned here: [...]

thnx u!