TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #24: Which FW to use

pstabno, 2023-08-09:
Hey - I wanted to install klipper from Your repo using SD card and one question - I noted only UART firmware image is updated 2 months ago. For SD should I use older one(7 months) from USB com?

TheFeralEngineer, 2023-08-09:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/24@
> github.com>

You can try it, but more than likely, you'll have to create your own bin
file to flash as newer Klipper versions require fresh bins to communicate
properly. I've provided information on how to do this in various YouTube
videos and all of the make menuconfig options are listed on the top of my
printer.cfg files. Also, be sure to disconnect the original screen before
flashing the bin, as it has been known to cause update issues if it remains
connected.

pstabno, 2023-08-09:
Screen is already disconnected and waiting for next step :) I wonder if first I should not try the older image and see how it goes. BTW - What are the risks of bricking mainboard with SD installment? if something fail it will just re-attempt to install with new image plugged?

TheFeralEngineer, 2023-08-09:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/24/1671189081

I've never had anyone brick theirs, so I'd say the chances are very slim.
All you have to do in the event that it doesn't work is try again. 🐱🐱

pstabno, 2023-08-09:
Ok all worked. Thanks a lot.

closed 2023-08-09 by pstabno