TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #25: Slicer gcode

HoudiniCerini, 2023-08-13:
I'm having a really hard time getting my abl to work properly! My screw tilt is within.01 max difference I will try to post a video everything is square and level so it really has me scratching my head.  I'm using the Neptune 3 plus. 

https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/assets/131235915/7d3cd4e7-97ed-46de-bcdd-646a71abb150

TheFeralEngineer, 2023-08-13:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/25@
> github.com>

What's the issue?

HoudiniCerini, 2023-08-13:
As you can see in the video the left side has perfect squish but the 2 squares to the right you can see the nozzle is getting further away from the build plate ☹️

TheFeralEngineer, 2023-08-13:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/25/1676171081

I'll take a look as soon as I can, I've seen replying via email so I didn't
see.

When you calibrate the mesh, you save, restart and activate the mesh,
correct?

HoudiniCerini, 2023-08-13:
Yes I ran screw tilt first got my bed level then g29 and saved to config. In my slicer gcode it does run a g29 before print as well same as on my ender 3 v2

Metalor696, 2023-08-15:
@HoudiniCerini apologies for butting my nose in. Just in case, make sure that your bed mesh is definitely loaded. I was having endless trouble getting my prints to adhere to the bed properly, went and built an enclosure thinking that was the problem. Turns out that whenever you restart you need to load the mesh again - I had no idea this was the case. As soon as I did that my levelling issues went away 😆

HoudiniCerini, 2023-08-15:
@Metalor696  you are all good I will take any help lol my slicer gcode calls for g29 and I can see that it says it saves and I can see that it is working on the control dashboard but there is definitely something going on

TheFeralEngineer, 2023-08-15:
Klipper doesn't know what G29 is, that's just a G code macro that is created to mask out the bed_mesh_calibrate command and does a data_save, which will save the data and restart klipper but it will not enable the mesh.

You should be focusing more on the M420 G code macro, which enables the mesh.

[gcode_macro M420]
description: Load the current mesh
gcode:
  BED_MESH_PROFILE LOAD=default

You can simply go to the bed mesh screen in fluidd or mainsail and enable it there, command M420 in your slicer (remove the G29 as it will constantly reboot klipper) or simply use the raw command of BED_MESH_LOAD=default in your slicer or send it through console.

No matter how you do it, the mesh won't be loaded until it is done.

HoudiniCerini, 2023-08-15:
@TheFeralEngineer I can see on mainsail where it is applying the bed mesh the bed mesh is saved under p1 
[image]

TheFeralEngineer, 2023-08-15:
email me your cfg files and your slicer start g code... ***@***.***

[Screenshot_20230814_233505_Samsung Internet]<https://user-images.githubusercontent.com/131235915/260621056-0395d499-b16d-4447-b0e1-0c84af7fbc31.jpg>

HoudiniCerini, 2023-08-15:
@TheFeralEngineer I'm not able to see your email address 😔

HoudiniCerini, 2023-08-15:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/25/1678383858

Will this work?

TheFeralEngineer, 2023-08-15:
my github username at g mail dot com