TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #27: Firmware compilation .config files

staehle, 2023-08-24:
Hi @TheFeralEngineer !

I'd like to compile my own Klipper firmware for my Neptune 3 Plus, however when I flash it, I can no longer communicate over USB, even though I have that config option on. Only your `ZNP_ROBIN_NANO.bin` file (for the USB version) posted here works for some reason. I think I have the config settings wrong (from `make menuconfig`). Not asking for support from you with that, but is it possible you could commit to this repo the `.config` files that you generated from `make menuconfig` to build both the UART and USB versions of the firmware? 

For example, in the `klipper` source, after you save your menuconfig, a hidden `.config` file is created with those settings you selected from menuconfig: 

```
staehle@machine ~/g/p/klipper $ ls -aF
./   config/  .git/     klippy/  out/      src/   .config      COPYING     Makefile
../  docs/    .github/  lib/     scripts/  test/  .config.old  .gitignore  README.md
```

Thanks!

TheFeralEngineer, 2023-08-24:
To clarify, you went into make menuconfig, you selected use extra low level settings, selected stm32f401, 32k bootloader, 8mhz crystal, pa10/pa9 for USB, save, exit and compile (make), rename to ZNP_ROBIN_NANO.bin and you're having issues? That's all I ever really have to do. Check the baudrate and make sure it's set to 2500000 or you have to specify what the baud is in your mcu section. Other than that, it should be pretty cut and dry. I don't have the .config file handy at the moment.

staehle, 2023-09-17:
Hey sorry I forgot to provide an update. Changing it to `PA10/PA9` was the solution! I compiled and flashed the binary, and it worked just fine! Thank you!

closed 2023-09-17 by staehle