TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #28: Neptune 3 pro UART

Badlok1, 2023-11-22:
Please help me, I am trying to connect BTT-PI to neptune 3 pro via UART. I tried your bin file and made my own too. I don't want to connect, although everything connects perfectly via USB.
In the file printer.cfg changed /dev/ttyS0, in the file boardenv "display" changed to "serial"
Connect to j17 on neptune and 6-8-10 pins on btt-pi.
What can be the problem?

TheFeralEngineer, 2023-11-22:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/28@
> github.com>

Did you enable uart in boardenv.txt?

Phil T.
The Feral Engineer

Badlok1, 2023-11-23:
> Did you enable uart in boardenv.txt?
> 
> Phil T.
> The Feral Engineer
> 
> 
> 
> 
> > Please help me, I am trying to connect BTT-PI to neptune 3 pro via UART. I
> > tried your bin file and made my own too. I don't want to connect, although
> > everything connects perfectly via USB.
> > In the file printer.cfg changed /dev/ttyS0, in the file boardenv "display"
> > changed to "serial"
> > Connect to j17 on neptune and 6-8-10 pins on btt-pi.
> > What can be the problem?
> >
> > ID:
> > <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/28@
> > github.com>

In the boardenv.txt file I changed the console parameter to "serial". Are there any other parameters related to uart?

Badlok1, 2023-11-24:
If anyone has the same problem, just try swapping the tx and rx on one end.

closed 2023-11-24 by Badlok1