TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #37: Neptune 3 Plus Board firmware files

mmucahitkaya, 2024-04-30:
Hello everyone,

I have a Neptune 3 plus and i am trying to set up klipper firmware with "kiuah". I have used to "Neptune 3 pro" firmware *.bin files but printer is stuck "update firmware" state. How can i build compatible  firmware for "Neptune 3 Plus"

TheFeralEngineer, 2024-04-30:
> Thank you so much. Also i have another question. My build settings like in
> screenshot. I think, maybe clock frequency is wrong but the other settings
> are correct. Could you please check my build settings?
>
> image.png (view on web)

8mhz crystal. The rest looks good.

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

mmucahitkaya, 2024-04-30:
8 Mhz crystal is not working :(

TheFeralEngineer, 2024-04-30:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/37/2086476954

That isn't enough information to troubleshoot your issue

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

martin-badin, 2024-04-30:
> You need to disconnect the screen when flashing the klipper binary. The screen is useless afterwards. Also, that binary is too old to run modern Klipper. Watch my YouTube videos on how to build your own. Phil T. The Feral Engineer [www.theferalengineer.com](http://www.theferalengineer.com) Catnip contributions 🐱🐱 https://paypal.me/timpsonmusicllc
> […](#)

Sorry for this question in this thread.

What do you mean "too old"? I have used your bin file. Should I rebuild firmware or klipper can update the firmware itself?

TheFeralEngineer, 2024-04-30:
> ID: </issues/37
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/37/2086548303

You need to build new firmware

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

mmucahitkaya, 2024-04-30:
When i try to build with this configuration, printer's firmware can not update. But when i applied second configuration, printer's firmware is successfully update. But this time i got this error "MCU 'mcu' configured for 84 Mhz but running at 85 Mhz!"

First config
[image]

Second config 
[image]

TheFeralEngineer, 2024-04-30:
you need to use 8mhz clock. I've used the same clock settings on dozens of machines and they have all worked properly. Usually the mcu error will just be a warning and will run properly, but you should be using 8mhz. If you look at your MCU, there is an 8mhz clock crystal installed on it.

image.png (view on web)<https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/assets/39586436/3fa1ca98-1385-4978-b38b-a92a1c1886be>

mmucahitkaya, 2024-05-01:
Thank you, i did it :)

closed 2024-05-01 by mmucahitkaya