TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #39: Unable to open file - klipper

br0cker, 2024-05-21:
Hi. 
When trying to print (sliced in Cura), an error pops up as in the title. 
If you change the line: 
```
[virtual_sdcard]
path: ~/printer_data/gcodes
```

Then everything prints without problems.

TheFeralEngineer, 2024-05-21:
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/39@
> github.com>

Log into your webui and check your moonraker log. It'll tell you what
extension your device is looking for vs. what is in your printer.cfg

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

br0cker, 2024-05-22:
``
[moonraker.log](https://github.com/TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/files/15401805/moonraker.log)

This?
 Repo has untracked source files: ['moonraker/components/timelapse.py']