TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers #43: Power and UART communication with PI over RJ11 port

maximusyellow, 2025-04-06:
Looking into the schematics of the main board I realized that the front RJ11 port used for the touchscreen provides a 5V output and UART communication on pins PA11 and PA12. In order to test the power, I took the cable out of the touchscreen, and managed to power the Pi (Orange Pi) using it. I am wondering how much work would it be to modify the firmware to allow UART communication between the PI and the printer to happen over the PA11 and PA12 pins instead of the PA9 and PA10 pins. Has this been done before?

I figured that instead of soldering a 4 Pin header onto J17 on the main board, all it would take to get the PI powered by the printer and have UART communication would be to utilize this RJ11 port that's no longer in use due to klipper firmware

Any suggestions and information would be greatly appreciated. Thank you

maximusyellow, 2025-04-14:
Looks like a PR for this has been merged into Klipper master branch https://github.com/Klipper3d/klipper/pull/6830, and is included in the v0.13 release

closed 2025-04-14 by maximusyellow

TheFeralEngineer, 2025-04-14:
> Klipper3d/klipper#6830 <https://github.com/Klipper3d/klipper/pull/6830>,
> <TheFeralEngineer/Klipper-for-Elegoo-Neptune-series-3D-Printers/issues/43/2802238269
> @github.com>

It's a pretty simple swap. Just build a new klipper binary with the
appropriate UART pins. My only concern is the current draw required by an
orangepi vs the original screen. I think a raspberry pi 4b pulls like 2
amps. I don't think the 5v on the elegoo board will like that for too long.
Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

maximusyellow, 2025-04-18:
Tested this out using a new Klipper binary, I would like to report this worked like a charm! Here are the menuconfig settings I used:

` Klipper Firmware Configuration
[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F401)  --->
    Bootloader offset (32KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (Serial (on USART6 PA12/PA11))  --->
(115200) Baud rate for serial port
[*] Optimize stepper code for 'step on both edges'
()  GPIO pins to set at micro-controller startup
`
In the case of my Orange Pi 3 LTS, UART3 ports needed to be enabled to allow for communication. 

In terms of powering the Orange Pi, I settled using a buck converter hooked up to the back of the printer (DC 24V 2.5A out) for now, but looking more into powering it from the RJ10 port. 

While I can run the Orange Pi from the 5V on the RJ10 port, I am not sure whether longevity wise it is a good idea. It seems that based on the schematics and BOM from Elegoo's repo, the [5V regulator ](https://www.mouser.ca/ProductDetail/Monolithic-Power-Systems-MPS/MP1584EN-LF-Z?qs=rC7bBWoQAAktQUA%2FYI26vA%3D%3D) can supply a maximum of 3 Amps. The potential current problem could be with how other devices were configured on the board. 3.3V devices (via another regulator), as well as all of the end stops, all appear to be using the same 5V output from the regulator. My inclination is to think that there is enough power available, but I would love additional insight into this investigation